POETSII/tinsel #109: Mis-matched send and receive sizes for tinsel send and hostlink receive

m8pple, 2021-05-28:
I was testing the POLite software simulator out on a few more things before sending a
pull request, and found that it actually fails on nhood-sync. The problem that is
detected in the simulation is that a message of type `NHoodMessage` is sent as
part of the finish handler:

https://github.com/POETSII/tinsel/blob/16bcb6db7b738b95bea6ebf9be46101633448860/apps/POLite/nhood-sync/NHood.h#L42-L44

which results in a message of type `PMessage<NHoodMessage>` getting sent out
through `tinselSend`:

https://github.com/POETSII/tinsel/blob/16bcb6db7b738b95bea6ebf9be46101633448860/include/POLite/PDevice.h#L418-L419

AFAICT the POLite softwitch then statically chooses this as the single message
size, as the only call to tinselSetLen is right at the start of the softswitch:

https://github.com/POETSII/tinsel/blob/16bcb6db7b738b95bea6ebf9be46101633448860/include/POLite/PDevice.h#L297-L298

However, on the host side it receives a maximum message size:

https://github.com/POETSII/tinsel/blob/16bcb6db7b738b95bea6ebf9be46101633448860/apps/POLite/nhood-sync/Run.cpp#L52-L54

The simulator flags this as a problem, as it considers it an error if the tinsel and host code don't
agree on the size of each message sent - it doesn't mind if the message size varies, but both sides
need to somehow agree on the granularity of each message transferred.

So I guess my issue is two-fold:

1. Is this a bug in nhood-sync? I think so, but it is less interesting than...
2. How is it working? If the host is only sent part of a message, why doesn't it block at that point?

My view was that the call to `hostLink::recv` should block, as there will be less than the
required number of flits in the channel, so the whole app should block on the SW side.

m8pple, 2021-05-28:
Oh, FFS... It just occurred to me after writing all that: we have more than one thread!

So you're only checking for completion here, you don't care about values. So if only
one device sent, then it would block. But because all the other devices push messages,
behind the first, you still get a maximum size message (albeit padded with flits
from other messages).

Does that sound right?

mn416, 2021-05-28:
Hi David,

A while ago Graeme requested that `HostLink::recv()` always consumes one message, regardless of the number of flits it contains.  Previously, it returned a single flit.  The problem was that in apps involving messages of more than one size, it is not always easy for the host to determine which size of message is being received (it would have to be encoded in the first flit's payload somehow).  So instead, we changed the granularity:

https://github.com/POETSII/tinsel/commit/92f2a4a826000f1fe0e17c5f0a987036c6809ae3#diff-7c7009239d44a8f66f7e4d931f3f4bcddd868b42b4b1c2f122d73ad9ebf80a89

Even thought the commit message mentions Imperial, I really should have tagged you and Jonny rather the silently merging!

Does this explain things? (I haven't looked in more detail at `nhood-sync` but can do if this answer is not sufficient.)

mn416, 2021-05-28:
I guess this doesn't really solve the problem mentioned; the host still needs to look at the message to know its size.  Maybe @heliosfa can recall more.  But hopefully the semantics is at least clear now?

heliosfa, 2021-05-28:
> Even thought the commit message mentions Imperial, I really should have tagged you and Jonny rather the silently merging!
> Maybe @heliosfa can recall more.

I recall Jonny being involved in the discussions at the time? I also recall that there was talk about keeping a backwards-compatible flit-based receive method if needed? but I think we decided it was not needed at the time.

> Previously, it returned a single flit. I think the problem was that in apps involving messages of more than one size, it is not always easy for the host to determine which size of message is being received (it would have to be encoded in the first flit's payload somehow). So instead, we changed the granularity:

Bingo, without introspecting the packet at the point of receipt, there was no way to know how many flits to get. This was at odds with how rx worked elsewhere in tinsel - you deal with packets, not flits - and had the potential to make life difficult for Motherships/supervisors. We pull out of hostlink as fast as possible and put into another buffer for processing, and it is only at that point we look at the headers we place in the first 8 bytes of a packet's payload to determine what the buffer is cast to. Introspecting earlier makes things a mess as fully generic code is having to deal with application-specific packet sizes.

What the change does is give you all of the flits associated with a single packet, padded to the length of a full packet. It is then up to the application to cast the buffer to the correct size/structure. There is no mixing of flits from different packets.